Proto-App/Proto-Android #149: Scope Time

YukarinnAM, 2022-06-26:
Hi there
This is a function request
I was using a capacitor try measuring how much amp hour on a range of voltages it has
And when using the scope i thought about a function that would really come it handy
and that would be that of being able to see how much time is it when you select two point
like this
[image]
And that it shows somewhere (maybe on the other side of the screen undeneath the scopes) how much is this time period
for measuring frecuencies and charge speed is really really usefull
And it feels like something that should be there

Thanks for the great job and reading this
Have a great day!

Awen2345, 2022-06-26:
did you forget the trigger lock on the oscilloscope , it's very important?

YukarinnAM, 2022-06-26:
Excuse me but... what is a trigger lock?

Awen2345, 2022-06-27:
Trigger controls allow you to stabilize repetitive waveforms and capture single-shot waveforms. The trigger makes repetitive waveforms appear static on the oscilloscope display by repeatedly displaying the same portion of the input signal
[image]

YukarinnAM, 2022-06-27:
Ohhh you are right, that would be incredibly usefull
and now that i remenber that exists on real osciloscopes

Proto-App, 2022-06-27:
@YukarinnAM 
Great idea! We will add this feature.

@Awen2345 
Could you make separate issue with your request?

Proto-App, 2022-07-26:
@YukarinnAM 
Implemented in version 1.10.0

closed 2022-08-01 by Proto-App