Proto-App/Proto-Android #150: Only 90 degree tracks

electron271, 2022-07-03:
Make tracks automatically be at only 90 degree angles, so you don't have to do it manually

Proto-App, 2022-07-05:
Probably duplicate, we have autorouter on our TODO list.

electron271, 2022-07-05:
I don't really mean autorouter, I mean that diagonal tracks won't happen, and like in KiCAD/Other software, it will automatically step to only 90 degree angles on the tracks, I'll send a example

electron271, 2022-07-05:
This should happen when routing:
[image]
Not this:
[image]

Maybe also allow 45 degree angles, and of course have a option to toggle it. I usually have to make straight angles manually in PROTO, which can be a bit annoying

Proto-App, 2022-07-10:
@electron271 
Ok thank you for explanation!

YukarinnAM, 2022-07-13:
Actually i think it would be great if we could choose the have that with a setting or something
because there are some circuit in which using diagonal wire connections makes them easier to visualize
so if both can stay it would be great

electron271, 2022-07-13:
Sort of what I meant, that is also something that can be handy having them off