Proto-App/Proto-Android #152: ALU Example submit

YukarinnAM, 2022-07-13:
Hello!
Maybe this could be too complex for a circuit example
but in case it isn't i'm here to submit this circuit
is an Arithmetic and Logic Unit (ALU for short)
Is one of the basic components that makes a computer work
the functionality of it is also quite simple
4 Full adders to make a 4bit adder
8 bits of data
4 for as initial
and 4 for sum
it uses a 2 complement system
where the last bit is negative
This reduces the max val you can have with 4 bits
but gives us a new min value on negatives, so the amounts of values you can mae remains 2^bits
One problem here would be the Adders, since they are part of the digital pack, but i could easily change them with ANDs, XORs & ORs Gates
But this post is more like a presentation of the circuit , since an example like this could be very educative on how a computer does basic math
[image]
[image]
[image]
[image]

The circuit itself is already big tho
This could also work as a future IC request maybe(?
since having an ALU component would allows us make some really powerfull circuits
But maybe for now that would be too much

Still, for now this is more like asking if this circuit could be an example
Since i'm not sure

PD: the last image is more like an aestetic part to explain the system used
PPD: The subtract is to activate the (A - B) mode

Thanks for your time reading this huge almost nonsense i just created

YukarinnAM, 2022-07-13:
Last Note: I made the circuit that "Sebastian Lague" Explained, he has a youtube channel
I think his mention is something appropriate to say

Proto-App, 2022-07-16:
@YukarinnAM 
Thank you for this example! It is great to see that our app can handle massive circuit like this :) Firstly we need to separate examples with restricted components (visible only for users with addon, in this case DIGITAL)

Proto-App, 2022-08-01:
@YukarinnAM 
We are currently working with examples, could you share your circuit with us?

YukarinnAM, 2022-08-02:
Of course
Here
https://www.protosimulator.com/share?circuit=ALU_1659435524838

Proto-App, 2022-09-03:
@YukarinnAM Example added in 1.11.0

closed 2022-09-03 by Proto-App