Proto-App/Proto-Android #155: Voltage sources without funktion in a circuit file

Sven130889, 2022-07-16:
Voltage sources without function in a circuit file.  Resetting and reloading the app and the circuit do not solve the problem.  Other circuits at my workplace are not affected by the problem.
[image]

Sven130889, 2022-07-17:
If I activate that, then it works.  If I deactivate it, it doesn't work again. ???
[image]

Proto-App, 2022-07-17:
Hi @Sven130889 
Could you share your circuit via mail?

Sven130889, 2022-07-17:
I have sent it to your email address.

Sven130889, 2022-07-17:
The name of the circuit is "Roulette"

Proto-App, 2022-07-17:
@Sven130889 Thank you, we will investigate this issue

YukarinnAM, 2022-07-18:
i have a similar problem
and it happens when you connect ideal wires in a loop

like this
[image]
it really seems to induce this kind of problems
it also doesn't have an alert , so is hard to tell when you messed up something

so the non ideal wire explains why it fixes it
It makes them resistors and not ideal wires

Proto-App, 2022-07-19:
@YukarinnAM @Sven130889 
Sounds like bug, we will check it.