Proto-App/Proto-Android #156: I have created some examples 😉

Sven130889, 2022-07-17:
Bistable multivibrator with dynamic input
[image]

3 Channel running light
[image]

LED Accelerometer
[image]

DC Motor controller PWM
[image]

Digital Counter
[image]

Adder IC
[image]

Rainbow color changer
[image]

Twilight switch
[image]

Relais Timer
[image]

Transformer with voltage regulation
[image]

Digital Cube
[image]
[image]
[image]

Sven130889, 2022-07-18:
https://www.protosimulator.com/share?circuit=3_Ch_Running_Light_1658164809994

https://www.protosimulator.com/share?circuit=Accelerometer_1658164827743

https://www.protosimulator.com/share?circuit=Bistbl_Multivibrator_1658164847909

https://www.protosimulator.com/share?circuit=DC_Motor_Control_1658164865310

https://www.protosimulator.com/share?circuit=Digital_Counter_1658164882077

https://www.protosimulator.com/share?circuit=Digital_Cube_1658164899598

https://www.protosimulator.com/share?circuit=Full_Adder_IC_1658164930403

https://www.protosimulator.com/share?circuit=Rainbow_Light_1658164965852

https://www.protosimulator.com/share?circuit=Relais_Timer_1658164995111

https://www.protosimulator.com/share?circuit=Twilight_Switch_1658165013644

https://www.protosimulator.com/share?circuit=Voltage_Regulator_1658165045182

Proto-App, 2022-07-19:
@Sven130889 
Thank you for such great examples! Could we use your nickname in app as an author of examples?

Sven130889, 2022-07-19:
Yes, for sure

Proto-App, 2022-09-03:
@Sven130889 
Examples added in 1.11.0