Proto-App/Proto-Android #157: Add new Examples

Awen2345, 2022-07-18:
1. 741 Inverting Amplifier
[image]
Link:https://www.protosimulator.com/share?circuit=741_Inverting_amp_1658110240975

2. 741 Non Inverting Amplifier
    [image]
Link:https://www.protosimulator.com/share?circuit=741_Non_Inverting_1658111176850

3.741 Slew Rate
[image]
Link:https://www.protosimulator.com/share?circuit=741_Slew_Rate_1658110292162

4. Square Wave Generator
[image]
Link:https://www.protosimulator.com/share?circuit=Square_Wave_Gen_1658110381289

5. Sine Wave Generator
 [image]
Link: https://www.protosimulator.com/share?circuit=Sine_Wave_Gen_1658110348612

6. Triangle Wave Generator
[image]
Link:https://www.protosimulator.com/share?circuit=Triangle_Wave_Gen_1658110417831

7. Sawtooth Generator
[image]
Link:https://www.protosimulator.com/share?circuit=Sawtooth_Gen_1658110164656

Awen2345, 2022-07-18:
8. CMOS VCO
[image]
Link:https://www.protosimulator.com/share?circuit=CMOS_VCO_1658115297933

9. AC 220v Dimmer
[image]
Link:https://www.protosimulator.com/share?circuit=Dimmer_1658115356831

Proto-App, 2022-07-19:
@Awen2345 
Thank you for stunning examples! Could we use your nickname in app as an author of examples?

Awen2345, 2022-07-28:
Yes i sure

Awen2345, 2022-08-12:
Hello @Proto-App 
i made a large circuit is IC Internal Schematic L3803 

[image]
[image]
[image]
I made a circuit from :
[image]
Link : https://www.protosimulator.com/share?circuit=L8038_1660304422135
Add my Examples to proto app

Proto-App, 2022-09-03:
@Awen2345 
Examples added in 1.11.0

closed 2022-11-01 by Proto-App