Proto-App/Proto-Android #160: Gates and Flip Flops

Awen2345, 2022-07-20:
[image]
Hello@Proto-App 
in this picture there are only 2 input logic gates, and flip flops there are no reset and preset pins, logic gates should have more than two inputs and flip flops have preset and reset pins, will you implement them in the next update?

Awen2345, 2022-07-28:
@Proto-App

Proto-App, 2022-07-28:
@Awen2345 
Thank you for suggestion. Probably we will add 4 and 8 input versions of each gate. In case of RST pin it will be available as an option because current models are simplified version.