Proto-App/Proto-Android #162: More Examples

Sven130889, 2022-07-26:
Full Substractor
[image]
https://www.protosimulator.com/share?circuit=Full_Substractor_1658858617471

Multiplexer
[image]
https://www.protosimulator.com/share?circuit=Multiplexer_inside_1658858664022

NUM Keyboard
[image]
https://www.protosimulator.com/share?circuit=NUM_Keyboard_1658858698873

PISO Register
[image]
https://www.protosimulator.com/share?circuit=PISO_Register_inside_1658858754393

Random Generator
[image]
https://www.protosimulator.com/share?circuit=Random_Generator_1658858797837

Roulette
[image]
https://www.protosimulator.com/share?circuit=Roulette_1658858841381

Traffic Light
[image]
https://www.protosimulator.com/share?circuit=Traffic_Light_1658858900527

Proto-App, 2022-07-26:
@Sven130889 
Thank you again! Great circuits!

Proto-App, 2022-10-13:
Implemented in v1.11.0

closed 2022-10-13 by Proto-App