Proto-App/Proto-Android #169: Add My Examples About Power Converter

Awen2345, 2022-09-03:
1. Buck Converter 
[image]
https://www.protosimulator.com/share?circuit=Buck_Converter_1662213301136

2. Boost Converter
[image]
https://www.protosimulator.com/share?circuit=Boost_Converter_1662213166671

3.Buck Boost Converter
[image]
https://www.protosimulator.com/share?circuit=Buck_Boost_Converter_1662213280509

4.SEPIC Converter
[image]
https://www.protosimulator.com/share?circuit=SEPIC_Converter_1662213412478

5.CUK Converter
[image]
https://www.protosimulator.com/share?circuit=Cuk_Converter_1662213392814

Proto-App, 2022-09-03:
@Awen2345 Thank you, we will add examples in next update.

Awen2345, 2022-09-19:
do you know about IC UC3843? @Proto-App

Proto-App, 2022-09-20:
Yes, we know about this IC

Awen2345, 2022-09-22:
Can you implement the UC3843 IC?

Proto-App, 2022-10-05:
@Awen2345 examples implemented in version 1.12.0

closed 2022-10-05 by Proto-App