Proto-App/Proto-Android #170: Cannot input capacitance with decimal.

wuyongzheng, 2022-09-13:
I want to have a 3.3uF capacitor, but I can only have 3 or 4. I can't even put two capacitors in parallel. It complaints C3 is shorted.

Proto-App, 2022-09-13:
@wuyongzheng 
Do you use input dialog to set capacitor value? By default capacitors does not have internal resistances and that is why you have message about shorted component. You can turn on "Wires resistance"  from settings panel (swipe from left), then you will be able to run simulation without issues.

wuyongzheng, 2022-09-14:
> 
Thanks for the response. I used the dialog. I mean click on the number and cannot type in the decimal point. Wire resistance completely distorted the simulation result. But I get the idea, I can add a small serial resistor to the capacitor.

Proto-App, 2022-09-15:
Yes, you can add small resistors. Ok, on values other than F you can not type ".", we will try to fix this design flaw.

Proto-App, 2023-04-08:
@wuyongzheng 
Implementd in version 1.18.0

closed 2023-04-08 by Proto-App