Proto-App/Proto-Android #174: PWM Controller Integrated Circuits

Awen2345, 2022-10-30:
Hi @Proto-App  I will give the datasheet link

1.SG3525
https://pdf1.alldatasheet.com/datasheet-pdf/view/23561/STMICROELECTRONICS/SG3525A.html

2.SG3524
https://pdf1.alldatasheet.com/datasheet-pdf/view/23560/STMICROELECTRONICS/SG3524.html

3.TL494
https://pdf1.alldatasheet.com/datasheet-pdf/view/12679/ONSEMI/TL494.html

4.UC3843
https://pdf1.alldatasheet.com/datasheet-pdf/view/12685/ONSEMI/UC3843B.html

5.UC3845
https://pdf1.alldatasheet.com/datasheet-pdf/view/25566/STMICROELECTRONICS/UC3845.html

In the datasheet link, there is an internal schematic, so I Need a lot of Integrated Circuits That is all and thank you

Proto-App, 2022-12-15:
@Awen2345 
Do you have already circuits made in PROTO?

Awen2345, 2023-04-21:
@Proto-App Hmmm not yet

Awen2345, 2023-04-21:
@Proto-App Hmmm not yet , it's better if there is Spice model import feature