Proto-App/Proto-Android #176: Enhancements

LucasBS1, 2022-11-11:
Enhancements

The app needs a LOT of enhancements:

1 - Lines and components are VERY hard to connect;
2 - Component properties (like resistance) are hard to change;
3 - The graph doesn't show the right unit (voltage or current) when more than one component is selected;
4 - Capacitor in parallel to an AC source is considered a short (as if there was no impedance).
5 - Can't see the power dissipated off potentiometers.
6 - AC source doesn't differentiate between peak and RMS.
7 - Can't disconnect a component (cut the lines): I have to either delete the component, or (sometimes works) rotate it
8 - It's hard to change the simulation speed and the plot scale.

Suggestions:
1A - Enlarge the touch-sensitive area that represents the pins of the components.

1B - The app deserves an auto route for the wires

2A - When I tap the center of a component (not the pins), the dial should pop up, displaying the last used variable (resistance, voltage). And somewhere, a button to select another variable.

2B - Plus, I get the stupid value input box every time I attempt to drag the slider. Make it come up only when tapping the center. And make this input box nearer the finger, with the value already highlighted.

3 - There could be three plots: one for current, one for RMS voltage, one for peak voltage, one for power... The user decides.
If the user sticks to a single plot, then one unit on the left, and another on the right. Both colored.

4 - Even ideal capacitors have impedance on AC

8 - Make the simulation speed change by pinching the plot.
Plus, there are many other variables around the app that could be controlled by gesture.

Proto-App, 2022-11-17:
Hello @LucasBS1 
1. We know that this might be improved, we have autorouter on our TODO list
2. It is quite easy, you need to tap on settings button and you will get circular dial with values.
3. It shows right unit because you can decide which unit is selected. We had discussion about multiple plots at one time and it can introduce additional complexity for beginners.
4. Our model has no resistance and from AC perspective start of the simulation is a shorted circuit. You can add small resistor or turn on wires resistance. 
5. Agree, we will fix it :)
6. Could you provide more details?
7. Do you mean cut wires between components?
8. You can change simulation speed from settings panel, we will introduce gesture from https://github.com/Proto-App/Proto-Android/issues/173
9. Yes, currently 3 are available
10. This feature is on our TODO list
11. It is ideal component
12. Agree, no flip or rotation. We will add flipp option because rotation is useless in this case.
13. We don't have any mathematical model of tertiary transformer

LucasBS1, 2022-11-17:
> 1. We know that this might be improved, we have autorouter on our TODO list

Cool. Please don't forget about my points 1A nd 1C

> 2. It is quite easy, you need to tap on settings button and you will get circular dial with values.

There is a step you forgot to mention between the circular dial and the settings button: I have to select the variable to be changed (name, resistance, voltage...)
That is NOT "quite easy". I'll attach a picture that shows how it could become so.

[image]

Plus, not to forget my points 2A and 2B

> 3. It shows right unit because you can decide which unit is selected. We had discussion about multiple plots at one time and it can introduce additional complexity for beginners.

1 - I do not see an option to change the unit to be displayed. All I see is the last unit I decided to plot. If I plot the voltage of R1, and then the current of R2, only Amps will be shown on the plot, not volts, from R1.

If the option IS there, and it's just ME that can't see it, it means the interface is counterintuitive.

2 - The users should decide what is complex or simple. People find insulting when decisions are made for them. Your competitor, EveryCircuit, makes it very simple: all you have to do is drag the plot down.

> 4. Our model has no resistance and from AC perspective start of the simulation is a shorted circuit. You can add small resistor or turn on wires resistance.

Suggestion: Add an option for an internal resistance (via, as you called on item 1, the settings button)

> 6. Could you provide more details?

I really could have been more clear...
And looking back, this "feature" would be quite redundant. But here we go anyways:
A sine wave has an RMS value and a peak value. Maybe, an option to choose one of these when adjusting the AC source.
I now judge this as redundant because I can see the peak value if I make a plot of this AC source.

> 7. Do you mean cut wires between components?

"Cut", I realized only now, is the term used by your competitor, EveryCircuit, to disconnect a component without deleting it. All it does is delete the wires connecting the terminals. Then I can move the components without dragging the wires along.

> 8. You can change simulation speed from settings panel, we will introduce gesture from [Swipe left and right for reducing and increasing simuilation speed #173]

Suggestion: swipe to control the speed, pinch to control the Time Step.
Although, both these variables should be a single one.

> 9. Yes, currently 3 are available

Well... 3 is nothing summed with nothing. In many apps there are at least 100 undo levels

> 11. It is ideal component

Well... Should be not. A typical relay coil uses some 20mA when 12V, for instance. And it has a kickback when switched off.
This doesn't have to be precise, but not having a limit affects the simulation.

> 13. We don't have any mathematical model of tertiary transformer

This doesn't have to be precise either. All we need is a ratio and a polarity, and MAYBE a coupling.
But the tertiary SHOULD be dettachable, movable, mirrorable, and rotatable