Proto-App/Proto-Android #177: Cannot open example because of digital pack

HyVong007, 2022-11-23:
I have bought the Bundle pack. But when i try to open the "Rainbow Light Effect" example, it says "The circuit contains unknown component: Digital pack" and the example is grey color. Please fix it !

Proto-App, 2022-11-23:
@HyVong007 
Could you send your order ID? It starts with GPA- prefix. Do you have access to all components during fresh simulation?

HyVong007, 2022-11-23:
> 
Transaction id: GPA.3303-9611-4504-96427

Now i can open "Rainbow light effect" example, it is strange, i dont know why !
I can use all circuit components from all packs.
I think this bug rarely happens but you should
check again, thank you !

[image]

Proto-App, 2022-11-24:
@HyVong007 
Great! Sometimes Google sends info about purchased products with some delay (Google is responsible for in app purchase payment process). We think this case occurred here (we had few reports with similar situation).

closed 2022-11-24 by Proto-App