Proto-App/Proto-Android #179: Potentiometer Slider Voltage Issue

kratoyd, 2022-11-26:
I have a potentiometer with its ends hooked up to a Vac source, and the slider hooked to an opamp input. The slider does not seem to read a valid voltage, instead just reading as noise.

foan, 2022-11-26:
I'm having trouble reproducing this issue. This may have been an issue with a net drawn over a terminal not connecting with said terminal. This is a seperate issue-- device terminals should not be the same symbol as net connections unless there is a net connection at that location to avoid ambiguity.

LucasBS1, 2022-11-29:
I may have found something similar, and thought I was crazy: 
OpAmps are supposed to have high input impedance, but some signals I connect on the ports are messed up by the OpAmp's presence.
When I remove the opamp, the signal is back to normal.

But do check if the ports are not just laying over a wire instead of a node. PROTO has thig big problem with wires.

Proto-App, 2022-12-03:
@foan 
Could you post sample circuit to reproduce this issue?

LucasBS1, 2022-12-03:
Of course:

https://www.protosimulator.com/share?circuit=BasicSupply_1670096701179