Proto-App/Proto-Android #188: Relay Enhancements/other suggestions

MaxChooch, 2022-12-24:
Firstly, I'd like to state that I am an auto mechanic delving heavily into electrical diagnostics. This app is amazing, but from an automotive perspective, I've discovered a few things this app is lacking.

1. When selecting a relay, it would be nice to be able to select between 4 pin or 5 pin relays as well as normally open or normally closed. 
2. Add a 3-phase automotive alternator option ( with variable rpm choices and selectable max amp output, WITH a case ground to the alternator's housing), and a 3-phase diode rectifier for turning it into usable d.c.
3. Switches, in the auto world we have pressure switches, temperature switches, and magnetic clutches (air conditioning compressors in cars have magnetic clutches to engage the compressor via a pressure switch). You've got "sensors" but no switches. I need more switches. The switches I am proposing merely need to complete a path to ground via a conditional variable. The variable for the pressure switch, for example, needs to go up to at least 500psi for modern air conditioning systems. A way to simulate the condition being met to trip the switch would be imperative as well.
4. There are numerous sensors in the apps components list. I don't understand how to use them. I've spent hours trying to figure it out. There seriously needs to be some sort of instructional for each component in this app. An app of this magnitude definitely needs an instructional/tutorial because I can find nothing online except one foreign guy's YouTube video that has no subtitles, and your promo video. It would go a long way with usability if there were examples of how to utilize certain components like the pressure sensors and temperature sensors..

You guys rock! Thanks for all your hard work.

Proto-App, 2023-01-18:
Hi @MaxChooch 
Thank you for very detailed description! Please check answers below:
1. Good idea, we will try to implement
2. DC power source has voltage option, it is ideal model and it has no amp rating. 
3. It is very hard without mathematical model or some kind of subcricuit. Do you have any source?
4. Same as above. To make any not typical component we need models or circuits.
5. Ok, you mean some dependent sources?
6. Correct, now we have only inductors.
7. Sensors read values from your device and produce output values. You can check accelerometer LED example, you can observe output values based on device's orientation.
8. We agree, we will implement it soon.
9. If you have issue like this, please share your circuit via mail proto.playstore@gmail.com. We will try to debug it.

LucasBS1, 2023-02-20:
To avoid the creation of infinite types of switches (magnetic, temperature, etc), there should be one dummy switch that is programmable.

After all, if I had a temperature switch, then the app would also need a heat source, and that heat source itself would need programming. So on...

MaxChooch, 2023-02-21:
So you're agreeing with my review that there needs to be more switches? I
don't think 3 or 4 styles of switches is "infinite" and there definitely
needs to be pressure switches. Like, for sure. So many things use pressure
switches it's ridiculous. All this would have to be, is a switch you could
click on that's labled pressure switch with an arbitrary value.  You
clicking on that switch, would be simulating it being activated by
pressure. That's all I'm asking for.. screw the device sensor activated
switches, who has a thermometer they can plug into their phone anyway? Give
me something I can freakin manually activate for crying out loud, how hard
is that to program?

LucasBS1, 2023-02-21:
Totally agree. We need something to emulate sensor switches (transducer). Don't necessarily agree that this transducer must read environment variables.

Right now, I use on/off switches, or push on/off switches. Like you. Those simple ones.
All we need (that is EASY for the developers to do) is a PROGRAMMABLE switch.
Then I can label that programmable thing to whatever I want, like "pressure switch"

MaxChooch, 2023-02-21:
Yes, I agree. Completely.

Proto-App, 2023-02-21:
@LucasBS1 
Please explain on example what  do you mean by saing programmable switch. It is obvious to you but not for everyone.

LucasBS1, 2023-02-21:
Easy to explain:

PROTO already has magnetism, gyro, and light sensors.
For those, we can simply attach a relay to it. Attaching a relay to it, it becomes a switch.
Basically, the switch we are talking about is a sensor attached to a relay.

(see also note 1 at the end)

There. Explained.

But those sensors are attached to our phone, and we don't need the variables that the phone gives us: we need variables that we will find on cars, industries, air conditioners, etc.

We could very well attach a voltage source to a relay, and call it switch. But what if we wanted that voltage to vary during the simulation ?

And to vary in a controlled manner

MaxChooch's suggestion of switch would require PROTO to simulate an environment with pressure, temperature, magnetism, humidity, etc
Here comes my suggestion, instead:
Let's imagine a black box. With how many  terminals we decide.
When we edit the content of the box, we can write a code on it. For instance:
- Open port number 5, 10 seconds after the simulation starts
OR
- When port number 2 reaches 15 volts, turn on port number 4
OR
- When port 8 is high, count to 200 and then make port 10 output 25 volts

So on.
We can simulate real life values with this.

I probably just described a microcontroller, which is why I suggested a microcontroller (twice: once in this thread and on another one of mine)

Note 1: Sensor+relay takes SPACE. It this regard, there's this suggestion, not made by me, that could help overcome it:
https://github.com/Proto-App/Proto-Android/issues/183

Proto-App, 2023-02-22:
@LucasBS1 
You simply described mcu because your description requires some kind of code execution. And it is not that easy for devs as you described because this solution needs language interpreter in runtime! But idea is great, we will think about it.

MaxChooch, 2023-02-23:
Proto Team, thank you for your continued interaction and willingness to
hear people out. I know sometimes we don't have the full picture of what it
takes to get things done, but I really appreciate your correspondence.

Troy

Proto-App, 2023-02-24:
Yes, we prefer to get as much info as we can :) 

@MaxChooch @LucasBS1 is there any possibility to give us your requirements about "programmable" switches (short summary)? Do you need only outputs, or maybe also inputs? how much? what about JavaScript as main language that controls logic inside switch?

LucasBS1, 2023-02-24:
I'm more convinced about the idea of a microcontroller than a programmable switch.
In the end, it's the same.

I already abducted Max's thread, so I'll leave switches for him to describe. But in my opinion, uC fulfills the needs of everyone.

We are accostumated with C language. It's the predominant language in microcontrollers.

But Java is good.
At least for the moment.
Maybe more options in the future.
The developers already have a lot on their hands

Microcontrollers have timers, AD/DA, PWM, interrupts, etc...
All we need is timers really. The rest PROTO already has.

And ports, of course
Inputs and outputs, lows and highs, maybe even analog levels

This is BIG, if you think about it. A microcontroller can replace so many other components...
And I don't know of any other software that can do this.

Proto-App, 2023-03-13:
@LucasBS1 @MaxChooch 
We will implement something between regular component and mcu. You will get N inputs, N outpus and code editor to write IC logic.

Proto-App, 2023-05-15:
@LucasBS1 @MaxChooch 
In version 1.20.0 yo ucan try JavaScript addon. It allows to write custom logic and drive outputs.

LucasBS1, 2023-05-15:
> 

Nice. Exaclty what the doctor ordered. 
I'll take some time to test it, since I'm familiar with C language