Proto-App/Proto-Android #193: own extras but can't access

stevemilkins, 2023-01-18:
Hi, I have purchased full extras pack but specifically can't see how to access enhanced scope. I've tried uninstalling and reinstalled the app. Still can't see how to use them. Am I missing something?App is supposed to be compatible with my device. I'm on the beta testing process 

Google account name is different so I can send it to you directly if you need it to confirm 
Thanks
Steve

Proto-App, 2023-01-18:
Hi

Do you have version 1.14.0 or 1.15.0? Do you have all other add-ons available?

stevemilkins, 2023-01-18:
Hi

Thanks for quick reply. Its 1.15.0  Have been looking during the day, I think I am being a bit dim, I hadn't worked out that there was a full screen mode!

Its a very usefull app, being on a tablet means I can test out some thoughts without having to have a PC with me. Well worth what it costs!

Steve

Proto-App, 2023-01-18:
Great!

closed 2023-01-18 by Proto-App