Proto-App/Proto-Android #194: Spdt issue

Jcriggs8, 2023-01-18:
The spdt analog switch is acting as a double active output instead of a single active output. When input is connected to a power source both outputs activate instead of just the selected output and activating switch dosent change output status

Jcriggs8, 2023-02-14:
Did the update and the spst and spdt analog switchs are still not working properly. The the outputs on both switchs all still stays active even when not selected output

Proto-App, 2023-02-19:
@Jcriggs8 
What do you mean by active output?

Jcriggs8, 2023-02-19:
The analog switchs keep a constant active output as soon as a power source
is connected.

Alanwew, 2023-09-05:
I really want you to add simulation of KA and SG3525 I really need it