Proto-App/Proto-Android #196: Buying bundle pack not working

GuVi171819, 2023-01-25:
I bougt the bundle pack but it doesn't work.

Proto-App, 2023-01-25:
Hi, could you give your order ID? It starts with GPA- prefix.

Proto-App, 2023-02-07:
@GuVi171819 
Please reopen if you have any issues with bundle pack access.

closed 2023-02-07 by Proto-App

GuVi171819, 2023-05-12:
still not working.
where can i find my purchase ID

Proto-App, 2023-05-12:
You can find it on your Gmail account. Google always sends confirmation with order ID.

GuVi171819, 2023-05-14:
bougt it again,
here's the order number:
GPA.3306-4071-1711-92648

Proto-App, 2023-05-14:
Thnak you, do you have acces to addon content?

Proto-App, 2023-05-14:
@GuVi171819 
As we can see payment is ok but your order is not verified by our application. Could you uninstall PROTO and install it once again from Google Play store?

GuVi171819, 2023-05-15:
NO,
OTHERWISE I WOULDN'T HAVE CONTACTED YOU.
I apologize for being rude.

GuVi171819, 2023-05-15:
my Email software mixed up the Mails.
It's still not working.

GuVi171819, 2023-05-15:
i already told you, but here:
GPA.3306-4071-1711-92648

GuVi171819, 2023-05-15:
mixed up again. I'm sorry