Proto-App/Proto-Android #202: Add New Examples About TTL Gates

Awen2345, 2023-02-08:
1. 74LS00
[image]

Link: https://www.protosimulator.com/share?circuit=74LS00_1675865471470

2. 74LS02
[image]

Link:
https://www.protosimulator.com/share?circuit=74LS02_1675865792750

3.74LS08
[image]

Link:https://www.protosimulator.com/share?circuit=74LS08_1675865522838

4. 74LS04
[image]

Link:
https://www.protosimulator.com/share?circuit=74LS04_1675865553631

5.74LS32
[image]

Link:
https://www.protosimulator.com/share?circuit=74lS32_1675865606227

Proto-App, 2023-02-08:
@Awen2345 
Thank you. We will add all of your examples in next release.

Proto-App, 2023-12-03:
@Awen2345 
Added in version 1.25.0

closed 2023-12-03 by Proto-App