Proto-App/Proto-Android #218: Scope Analyze Function

standardE, 2023-04-04:
I'm having a very hard time figuring out how to use the scope. Whenever I try to analyze more than one attribute at a time it seems that the second attribute gets inverted. For example, in this circuit below, I expect current and voltage to rise together and fall together as the power supply voltage changes. But instead, one trace always inverts. Am I not using something correctly?
[image]

Proto-App, 2023-04-04:
Hi @standardE 
You use it correctly. Scope shows opposite values because we have to measure somehow current via wire. In this example current has negative value because PROTO can not decide which point of a wire is a "reference" point. You can delete wire from example and place new one from R3 (first point) to R4. First point is a refrenece point to measure and you will get positive value that matches voltage changes.

standardE, 2023-04-06:
Ahh! Ok, I will mess around with that. I figured it was just some little detail that I didn't know. Thanks!