Proto-App/Proto-Android #219: Intermittent Stimulation Issues

standardE, 2023-04-06:
I am experiencing intermittent issues in which circuits will be fine to act illogically. For example, a resistor value increase would normally increase a given voltage drop and a decrease would decrease it; but I have experienced the exact opposite. In situations like this the simulation stops working properly and I need to reset it to get the sim to operate properly again. This latch circuit below is the most recent example. At times hitting the button a few times works (along with the voltage drop logic) and thereafter will stop working no matter what tweaks I make to it. I verified the basic design and values with another app (also attached) and it works flawlessly every time with verry little adjustment. 
[image]
[image]

standardE, 2023-04-06:
Also, when voltage drop logic messes up, so does current logic.

Proto-App, 2023-04-23:
@standardE 
Thank you for report. We will investigate both issues. Coud you share your circuit via mail? You can send it to us: proto.playstore@gmail.com

standardE, 2023-04-23:
Unfortunately, I cannot sent you anymore than what I posted. I hade to
reimage my phone and because there is apparently no sort of cloud backup
for your app I lost my entire workspace. I hope that the images I provided
are enough if reproduced.