Proto-App/Proto-Android #221: add VCC and GND to OPAMPs

willbrs1285, 2023-04-11:
I think the addition of OPAMPs VCC and GND would be nice as I'm confused where current goes to and comes from and it would be more customisable as it seemingly goes to and comes from the negative of a supply and they act weirdly when a ground is not present in the workspace/not connected directly to the circuit in which the OPAMP is and everything messes up sometimes, so It would be nice to have such customisability.

standardE, 2023-04-12:
I'm so glad to hear someone else wantint this item for the same reasons I have. To make a split rail and know exactly what is ground and what is virtual ground to the OpAmps would be SO nice! The way it's set up now seems to be assumed which largely defeats the purpose of the app. Other comparable apps have this same problem. PROTO would be a leap ahead of the competition if they had a fully pinned out OpAmp; and to go a step further how about one with bias compensation!

uopeppe, 2023-04-20:
I think it's better now, because you can increase/decrease the Voltage Output, which is the supply voltage from "Edit Parameters" that results most versatile than the external pins.

Weslley875, 2023-07-16:
Cara eu gosto muito desse simulador só deixa a desejar que os opamps não funciona muito bem e séria bem legal se conseguisse melhorar mais ele colocando vcc e gnd nas alimentação.