Proto-App/Proto-Android #222: Negative Linear Regulator

standardE, 2023-04-12:
A neat component would be negative Linear Regulators...

Proto-App, 2023-04-18:
@standardE 
Do you mean something like 7805?

standardE, 2023-04-19:
Hey! Thanks for they reply!

I was thinking about something like the 7905 (79xx series). It's the same
thing as the 78xx series except the 78xx series source voltage while the
79xx series sink it. 79's are basically a negative voltage regulator.

Also, I read the other person's request for a fully pinned out OpAmp
(including Vcc, Vee and possibly even null pins). It would be so cool to
have a 'real world' version of an OpAmp to mess around with ; say, an
LM741. It would be so cool to simulate circuits with a real world amp and
set the null. Are you guys thinking about that option?

Proto-App, 2023-04-19:
Yes, we have real opamps on our list (741, 358)

standardE, 2023-04-20:
That's fantastic! I appreciate the work that you guys put into this app. I
look forward to future updates.

standardE, 2023-04-22:
I had to reimage my phone, where's my saved circuits? Where is this stuff
baced up to?

Proto-App, 2023-04-23:
Now you can not export whole workspace, you can try to share each item from your workspace but it can be uncomfortable. We will implement workspace export.

standardE, 2023-04-23:
Ya, some sort of backup linked with my account would be a massive help.
Thanks!