Proto-App/Proto-Android #223: Circuit item description unreadable on screens with rounded corners

Piotrassin, 2023-04-15:
Hello,
I am using a Samsung Galaxy S9 which has rounder corners and it interferes with the data displayed I the top left corner upon clicking on any circuit item.
Steps to reproduce.
1. Start a new circuit or load an example
2. Go to settings panel and make sure to have "show details" enabled in the settings. In polish translation it's labeled "Pokaż opis" and I can't change the app language so I am guessing the English name. 
3. Tap on any item on the circuit
4. The first two lines of the details is interfering with the rounder corner. The first line is sometimes unreadable as the text is off screen. 
5. Visualisation picture below. 
I modified the screenshot to visualize the problem as the screen in rounded, but the screenshot is not and normal screenshot shows the details properly. 

[image]

Proto-App, 2023-04-15:
Hello @Piotrassin 
Thank you for very detailed description. We will check possible solutions, but we are not sure that there is any Android API to check if device has rounded corners.

Piotrassin, 2023-04-17:
Thanks! Possibly a slight margin from the screen left and top borders will fix the problem? It's not that elegant, but it should work :D

Proto-App, 2023-04-18:
Indeed it should work but we don't want to apply small margin on every device with regular corners. At least we will add your device and handle it differently.

ricKapali, 2023-10-25:
...having the same issue w/Samsung Galaxy Fold 4.  Any chance of adding the Fold 4 to the "exception list"?
Thx in advance!

[image]

Proto-App, 2023-10-26:
Hello @ricKapali 
Yes, it is hard to believe that Android does not provide any API to get corner size 😐

Piotrassin, 2023-10-26:
> Indeed it should work but we don't want to apply small margin on every device with regular corners. At least we will add your device and handle it differently.

I agree. Every device is different and may have a different corner radius hence making an exception with different margin rules for each one is rather cumbersome, but what if there was a static margin in place? 
Basically the same thing as the other UI buttons have. 
The text is small that adding a margin wouldn't obstruct the view, but if the margin was large enough it would solve the issue for almost all of the devices.

ricKapali, 2023-11-03:
> @Piotrassin <https://github.com/Piotrassin> @ricKapali
> Could you share model number from phone settings?
> [image:
> samsung-galaxy-find-imei-serial-number-model-number-phone-imei-serial-model-number-displayed]
> <https://user-images.githubusercontent.com/63373346/280342911-a0fee9df-729a-4968-b083-12d449a15f78.jpg>

Absolutely...thank you(!) in advance, Rich.

SM-F936U1

Proto-App, 2023-12-03:
@ricKapali @Piotrassin 
Implemented in version 1.25.0