Proto-App/Proto-Android #226: Mosfet

Ranghd95, 2023-04-20:
doesn't have any real mosfet-like properties

Awen2345, 2023-04-21:
Hmm there is no mosfer parameter option, I'm a bit disappointed

Awen2345, 2023-04-21:
Hmm there is no mosfet parameter option, I'm a bit disappointed

Proto-App, 2023-04-23:
@Ranghd95 
Yes, we know that MOSFET settings are quite poor. We will try to implement at least  voltage threshold.

Awen2345, 2023-04-23:
@Proto-App how about Length and Width Mosfet parameters?

Ranghd95, 2023-04-28:
> @Ranghd95 Yes, we know that MOSFET settings are quite poor. We will try to implement at least voltage threshold.

Yep, it's good to have R(DS) on

> @Ranghd95 Yes, we know that MOSFET settings are quite poor. We will try to implement at least voltage threshold.

yes it would be better if the mosfet has Rds on and Voltage threshold