Proto-App/Proto-Android #230: No response from the simulation until I turn on wire resistance, then Lags very badly

Paulmsam, 2023-04-23:
I have this issue with this simulation. It runs when the wire resistance is turned on. But then lags incredibly. I have turned off and on all options, rebooted my phone (Samsung galaxy s23 ultra), and created a copy, same thing with the copy. It works if creating a new simple circuit. Any advice?  

[image]

Paulmsam, 2023-04-23:
it was working perfectly, but then suddenly it started lagging

Proto-App, 2023-04-24:
Hi @Paulmsam 
Is there any chance to share this circuit with us? You can export it and share via mail: proto.playstore@gmail.com

Paulmsam, 2023-04-24:
Hi. Thanks for the reply. I have sent you a mail.