Proto-App/Proto-Android #242: Share circuit link brings to app download page

Whosdeez, 2023-05-24:
Even when the app is downloaded, clicking on a link to a shared circuit doesn't have any effect other than bringing up the Play Store app download button

Jurgen2003, 2023-05-24:
That's a phone setting. The proto app is not standard for that link 
[image]

Whosdeez, 2023-05-24:
@Proto-App I'm using a Samsung S10 Edge running Android 13. A friend of mine is also running a version of Android.
Thanks @Jurgen2003 your solution works.
It would be great if www.protosimulator.com web links were activated as default.
Though, note that pasting the link in a browser still doesn't redirect to the app. It has to be CLICKED on (for example, in a discord message, Word document or e-mail) in order to open the app and add the circuit automatically.

Whosdeez, 2023-05-24:
[image]
[image]

Jurgen2003, 2023-05-24:
Dear @Whosdeez I had the same problem and found out about this. I read your problem and recognized it right away nice that it works for you too 

Greetings Jurgen

Proto-App, 2023-05-24:
@Jurgen2003 Thank you for help! Interesting because on our Motorola G82 this setting is turned on by default :thinking: We assume that this might be some kind of customization on Samsung devices or maybe this is related to Android 13. Our Moto has Android 12. We will investigate it  :+1:

Proto-App, 2023-09-05:
It is probably customized by device provider and we can not fix it.

closed 2023-09-05 by Proto-App