Proto-App/Proto-Android #243: can someone explain to me how such a java script ic works

Jurgen2003, 2023-05-24:
can someone explain to me how such a java script ic works i find it a tricky one and can't figure it out

Proto-App, 2023-05-24:
Hello @Jurgen2003 
Each Script IC has sample code. You can place it on simulation screen and connect for example standard DC source. It should react and driver one of the outputs to high state. Of course you can change code that is responsible by this behavior. Simply tap on component, tap "Settings" button and select "Script" from dialog window. This action will redirect you to code editor.

Jurgen2003, 2023-05-25:
hey thanks for your response i saw it that the javascipt is the control but if you want to change it and program it yourself how does that work don't get anything from it get error messages all the time and it doesn't work is the proto app by the way a language adjustable speak more easily dutch how about that ?

Bushmills, 2023-05-25:
(Automatic) translation tends to yield quirkier results which may make the program actually harder to use than English localisation.

Jurgen2003, 2023-05-25:
thanks for the answer but so it is possible to set the language ? if so do you have an idea how and where ? @Bushmills

Proto-App, 2023-05-25:
@Jurgen2003 On Android 13 you can change language per each application, please check [https://android-developers.googleblog.com/2022/11/per-app-language-preferences-part-1.html ](https://android-developers.googleblog.com/2022/11/per-app-language-preferences-part-1.html )

Jurgen2003, 2023-05-26:
@Proto-App proto is not listed there

Proto-App, 2023-05-26:
@Jurgen2003 Ok, you are right. PROTO is not prepared for it now.... We will add possibility to change language at least from system setting in next release.

Jurgen2003, 2023-05-28:
@Proto-App  
With javascript I am not so familiar but can java. Make calculations for example voltage input 1 -volt input 2. Is voltage output 1 or 1 +2 is output or 1×2 so there is a calculation. Of the variable on input 1 and 2 one. Variable makes output 1 

Please if possible.I would like to understand how you write that script 

Regards jurgen

Proto-App, 2023-06-30:
@Jurgen2003 
Implemented in version 1.21.0

closed 2023-06-30 by Proto-App