Proto-App/Proto-Android #244: I bougt the bundle pack but it doesn't work.

LucasPaivaSilva, 2023-05-25:
I bougt the bundle pack but it doesn't 

[image]
[image]

Proto-App, 2023-05-25:
Hi @LucasPaivaSilva 
Could you uninstall and then install PROTO again? Very strange issue, we can't reproduce it on our devices 😐

LucasPaivaSilva, 2023-05-25:
Sure, i did that,  same problem Enviando Screen_Recording_20230525_153139_P R O T O.mp4 …

Proto-App, 2023-05-25:
@LucasPaivaSilva 
You don't have this addon because Google validation is unsuccessful  ?! Could you clear app data once again and run PROTO?

Proto-App, 2023-05-25:
If it won't help then we will return your payment.

LucasPaivaSilva, 2023-05-25:
You mean my payment validation (the one with my ID), it was just confimed now, but i still cannot use the addon. I tried cleaning the local cache and reinstaling multiple times. One this, i noticed that everytime i download from playstore it does come with my circuits, do you keep any cache/data in the cloud? how could i clean that?

LucasPaivaSilva, 2023-05-25:
btw, some extra information
device: Galaxy Tab S6 Lite
android version: 13 (5.1 on OneUI)

Proto-App, 2023-05-25:
@LucasPaivaSilva 
Hmm, that is very strange that you have previous circuits. Each "clean" installation should have always empty workspace, maybe this is the key related to your issue. We think that PROTO has previous state of this addon but we don't have any Samsung device with Android 13 :disappointed:  ... We checked logs from our server and we can't see request from your device with your order ID. Do you execute exact steps like in [https://www.samsung.com/us/support/answer/ANS00077611/](https://www.samsung.com/us/support/answer/ANS00077611/) ?

Proto-App, 2023-05-25:
By the way we don't use any cache, typically we save circuits in app standard directory that is available only for our application. It should be removed when you clear app's data.

Proto-App, 2023-05-26:
@LucasPaivaSilva any news in your case?

LucasPaivaSilva, 2023-05-26:
ow, i am sorry, i tried again a few times, in one try i opened the app after cleaning the data/cache to see if there was some circuit remaining (tried a few times before too) and then the app crashed. After opening again all the addons show as avaliable, so i dont have a problem anymore

LucasPaivaSilva, 2023-05-26:
but if it is of your interest i could help to search on this, because there is a bug somewhere (maybe on android itself or samsung cache management)
Thanks for the help! i am going to change my review on playstore, i loved the app, happy that i can use the addons now! c:

Proto-App, 2023-05-27:
@LucasPaivaSilva 
Great news! We are almost 100% sure that this issue was related to Android/Samsung and it might be some kind of cache. We don't have any Smasung device but maybe we should have one :)

closed 2023-05-27 by Proto-App