Proto-App/Proto-Android #245: Suggestion: PCB design

LucasBS1, 2023-05-27:
After you fix this [issue](https://github.com/Proto-App/Proto-Android/issues/176), and this [issue](https://github.com/Proto-App/Proto-Android/issues/182) 

I suggest that PROTO creates another section, or even an separate app, to **design PCB**s

Don't believe it's hard: resizeable round pads, square pads... chips with 8 legs, resizable capacitors and resistors, and lines connecting all of them.

I'd do this app myself if I knew more java than I thought I knew

And get rich in the process, because the only decent app tha used to do that has just been [retired](https://play.google.com/store/apps/details?id=com.theophrast.droidpcb)

Proto-App, 2023-05-27:
Hi @LucasBS1 
In near future we will release survey in PROTO to find out priorities. PCB design will be one of them and probably should be separate app where you can assign packages to components (like in KiCAD) and then start with PCB design. Next idea is to release online library.

LucasBS1, 2023-10-05:
Any update about this one ?
Temporarily, at least, I would suggest a simpler version without the library, just a very generic package.

It would be the backbone of the full app.
The light version could provide revenue to finance the full version