Proto-App/Proto-Android #246: how to write a script

Jurgen2003, 2023-05-30:
@Proto-App
With javascript I am not so familiar but can java. Make calculations for example voltage input 1 -volt input 2. Is voltage output 1 or 1 +2 is output or 1×2 so there is a calculation. Of the variable on input 1 and 2 one. Variable makes output 1

can you program the operation of an opamp in a javascript ic
?
Please if possible.I would like to understand how you write that script

Regards jurgen

Proto-App, 2023-05-30:
@Jurgen2003 
What kind of logic do you need? We can create small script and explain some parts. Regarding errors, all messages are always in English.

Proto-App, 2023-06-05:
@Jurgen2003 
Sorry but we don't understand your description. Could you make it with your native language?

Jurgen2003, 2023-06-05:
Ik wil werken met analoge/ variable voltage 

Dus met javascript. Het meten/lezen van een in komend voltage. En die. Beïnvloeden door een analoge /variable voltage op een andere ingang en dat als uitgang   

[image]

[image]

![Screenshot_20230605_125047_P R O T O.jpg](https://github.[image]

4609-9329-06feb60c6125)

![Uploading Screenshot_20230605_12![Screenshot_20230605_125108_P R O T O.jpg](https://github.com/Proto-App/Proto-Android/assets/128260279/f0ac27c8-2c67-[image]

ng Screenshot_20230605_125108_P R O T O.jpg …]()

[image]

[image]

Voorbeeld. Op ingang 1 staat 5 volt ( tussen de 0 en 10 op. Ingang 2 staat. 3 volt daarmee rekenen of. De 2 plus 5 of de 2 - 5 of keer. En naar mee een uitgang maken 

I want to work with analog/variable voltage 

So with javascript. Measuring/reading an incoming voltage. And that. Influenced by an analog /variable voltage on another input and that as an output 

Example. On input 1 is 5 volts ( between 0 and 10 on. Input 2 states. 3 volts with that count or. The 2 plus 5 or the 2 - 5 or times. And to along make an output. 

See the photos as examples. So I want to be able to regulate but the voltage on the input and outputs is variable

Proto-App, 2023-06-05:
@Jurgen2003 
On your screenshots you have dependent sources. If you want to use java script then use Script IC, it looks like below:

[image]

Jurgen2003, 2023-06-05:
[image]

 Yes I know that to with those sources can make the calculation input a +b a-b.   A×b.    I want to make such calculation with inputs make in java script and send an output there but how. Do you make a script that reads such an input on voltage and calculate with that  
 So following screen. How to make a javascript that controls the 5×2 or 5-2 or 5+2.

Proto-App, 2023-06-05:
@Jurgen2003 
Ok, now we understand your requirements :) So you have to choose script IC  (2 in - 2 out), tap on "settings" button and choose "Script" option. Then paste following code in editor and click "Save" button:

```js
function setupTerminals() {
    out0.setLowVoltage(0.0); //Can be removed, but you can set voltage in low state (not used here)
    out0.setHighVoltage(in0.getVoltage() + in1.getVoltage()); // out0 voltage = input 0 + input 1
    out1.setLowVoltage(0.0); //Can be removed, but you can set voltage in low state (not used here)
    out1.setHighVoltage(in0.getVoltage() - in1.getVoltage()); // out1 voltage = input 0 - input 1
}

function driveOutputs() {
    //turn output to high state 
    out0.setState(true);
    out1.setState(true);
}

//code loop executes both functions
setupTerminals();
driveOutputs();```

Proto-App, 2023-06-05:
@Jurgen2003 
[image]

Jurgen2003, 2023-06-06:
thanks for the help and the script i now understand how to make the script measure the voltage and calculate with 2 inputs 
 i programmed the 4 in 4 out and 8 in 8 out as well javascript is handy to work with but one more question  

 can you also add a fixed value to the input and output which can be calculated further see pictures 

 for example input .0 x ( a value ) + input.1 x ( a value ) or input.0 - input.1 x (a value ) the x   
  
[image]
[image]
[image]

Jurgen2003, 2023-06-07:
Is it possible, after calculating 2 inputs and the corresponding.  Increase and decrease factor (I managed to do that) to increase or decrease the output by a factor

Jurgen2003, 2023-06-09:
dear @Proto-App  i am going to understand java script and program whole circuits in it and it works a only i want to amplify some outputs for example input1 is 1v - input2 500mv (0.5v ) so output is 950mv (0.95v) where should you type the x10 to get 9.5 out so a magnified factor on the output  
 greetings jurgen

Proto-App, 2023-06-09:
Hi @Jurgen2003 
You should change following line in code:
```js
out1.setHighVoltage(in0.getVoltage() - in1.getVoltage()); // out1 voltage = input 0 - input 1
```
to:
```js
out1.setHighVoltage((in0.getVoltage() - in1.getVoltage()) * 10.0); // out1 voltage = (input 0 - input 1) * 10
```
or you can extract variable and code will be more readable: 
```js
let outputVoltage = (in0.getVoltage() - in1.getVoltage()) * 10.0;
out1.setHighVoltage(outputVoltage);
```

Jurgen2003, 2023-06-09:
Thanks@Proto-App I will try to increase an output after a calculation is it also possible to limit an output so minimum voltage and maximum voltage there between a control value  
 So min 2 v. And max 8 v on the output. And the input.  From 0 to.  10 have. 
 Greetings Jurgen

Proto-App, 2023-06-09:
Sure, you can make own "clamp" function and limit input/output in custom range, good resource to start:
[https://webtips.dev/webtips/javascript/how-to-clamp-numbers-in-javascript](https://webtips.dev/webtips/javascript/how-to-clamp-numbers-in-javascript)

Jurgen2003, 2023-06-09:
@Proto-App glad it can. It would be very helpful if I could 
 I have already read and polled that site a few times. Keep getting errors.    How. Which. Code is. For min and max limit.

Jurgen2003, 2023-06-12:
I manage to program entire scheme with multiple calculations. With multiple inputs providing information. And with that control the outputs  
But. I would also with the info on inputs. Want to limit a minimum voltage and maximum voltage I know with the clamp function it can but every time programming. Do I get errors joe works that clamp function  
 Thanks in advance

Proto-App, 2023-06-13:
@Jurgen2003 
Please paste your code here and write where you want to limit your input/output.

Jurgen2003, 2023-06-13:
Dear .....     
 I want to begin to understand 
  Suppose you have this setup how it ic.  Input 0 +input 1 = output 0. So. Out0.setHighVoltage in0.getVoltage ()+in1.getVoltage ()) 
 But how do you tell the ic that you want to limit with inputs 3 and 4    
 So. Input 3 minimum voltage and input 4 maximum input 1 to 4 are controlled  
   It is an example to see and understand the script 
[image]

Proto-App, 2023-06-15:
@Jurgen2003 
It is quite easy:

```js
function clamp(num, min, max) {
     return Math.min(Math.max(num, min), max);
}

function setupTerminals() {
    out0.setLowVoltage(0.0); //Can be removed, but you can set voltage in low state (not used here)
    let outVoltage = clamp(in0.getVoltage() + in1.getVoltage(), 2, 12)    //clamp out voltage between 2v - 12v
    out0.setHighVoltage(outVoltage);
    out1.setLowVoltage(0.0); //Can be removed, but you can set voltage in low state (not used here)
    out1.setHighVoltage(in0.getVoltage() - in1.getVoltage()); // out1 voltage = input 0 - input 1
}

function driveOutputs() {
    //turn output to high state 
    out0.setState(true);
    out1.setState(true);
}

//code loop executes both functions
setupTerminals();
driveOutputs();
```

Jurgen2003, 2023-06-16:
Thanks@Proto-App  I'm going to work on it

closed 2023-06-30 by Proto-App