Proto-App/Proto-Android #263: Bug

Jurgen2003, 2023-07-16:
Dear proto 
 There are some problems in this version so import is not possible comes this message. On 4g on wifi in the previous version it did work on these networks. Furthermore I have sometimes a circuit suddenly has no voltage and does not come back how is this possible? 
 Greetings Jurgen 
[image]

Proto-App, 2023-07-16:
Hello @Jurgen2003 
Strange, we did not change anything from last version related to connectivity. Could you check Addons tab under 4G network?

Jurgen2003, 2023-07-16:
Hello @ so as I said it is with 4g and wifi .  Greetings jurgen

Proto-App, 2023-07-16:
@Jurgen2003 Could you check Addons tab under 4G/wifi network?

Proto-App, 2023-07-17:
@Jurgen2003 we can't reproduce it on our side.

Jurgen2003, 2023-07-17:
@Proto-App Yes, of course I understand, but I find it strange and I want to tell you. 
  Greets jurgen

Ryan27777, 2023-07-25:
I have bug

Ryan27777, 2023-07-25:
I'm confused why the 4000 cmos IC hasn't opened even though I already bought it

Ryan27777, 2023-07-25:
[image]

closed 2023-09-05 by Proto-App