Proto-App/Proto-Android #265: Finding components and putting them is a hassle

thesunRider, 2023-07-17:
It would be very helpful if there was a ui such that the components are in a slider docked to the top and clicking on the component puts it in the middle of the screen,

Similar to A UI like everycircuit app's component placer.

It is currently very distracting to open the component selector everytime to place components,

Apart from that.The app is amazing, loving it. Hope you add this simple feature to it, Will be a huge improvement in using the app efficiently.

thesunRider, 2023-07-17:
If not the entire library,Adding a slider with the basic components, ground, resistor cap transistor, dc source etc can be very helpful

Proto-App, 2023-07-17:
Hello @thesunRider, thank you for suggestion. We will try to improve it or at least we can add some option to change this behavior.

LucasBS1, 2023-10-05:
Maybe the current standard toolbar, and a collapsed version of it