Proto-App/Proto-Android #268: Question about state in javascript

Jurgen2003, 2023-07-29:
Best @Proto-App   you at. Javascript at the state [out0.setState(in0.getVoltage()>0); ] also assign multiple inputs  
 For example. Inputs0 must be greater than 0 and inputs1 must also be greater than 0 but not added together    
 So. out0.setState(in0.getVoltage()>0+in1.getVoltage()>0 ) so iest this is possible. If yes how does it work  
 Greets jurgen

Proto-App, 2023-07-29:
Hello @Jurgen2003 
You need to make logic condition like this:
`out0.setState(in0.getVoltage() > 0 && in1.getVoltage() > 0 )`

Jurgen2003, 2023-07-29:
Best. @Proto-App javascript is getting clearer and works nice in circuits and with javascript ic too. The circuits smaller informate thanks for your quick and clear response and thanks again for the app language in the last update 
  Greetings jurgen

Proto-App, 2023-07-30:
That's great that JS is useful for you! Best Regards

closed 2023-07-30 by Proto-App