Proto-App/Proto-Android #270: New example: switching power supply

zom-betatester, 2023-08-08:
Hi, proto. I have made a new example, It is a universal power supply running ac to dc LV.

Let me tell you about the psu:

The power supply uses a relay to power the transformer and the secondary.
Relay turns on if the secondary voltage is too low and turns off if the secondary voltage is 
 good enough or if the voltage is too high. The relay has a variable switching, the feedback 
 comes from the current sensing resistor with a diode for reverse current protection for the 
 secondary capacitor.
The input voltage is universal (90V-265V) [might be compatible with 266V-543V]
The input frequency is universal (5Hz-158,000Hz) [might be compatible with higher freq]
The output voltage can be adjusted (0V-100V) [regardless of wattage]

To change the output voltage, change the value of the current sensing resistor. It is located in the relay coil circuit.
Sorry for a lot of information, but its good to know so you don't mess up.

Here's the image file:
[image]

Please note: My timestep is set to 100uS, higher or lower may make the switching psu malfunction.

In the future i'll improve it to be more efficient, the higher the usage at the secondary output rhe more efficient the switching psu is.

Proto-App, 2023-08-08:
Hi @ghost 
Thank you for this great example, we will add it in next release. Best Regards

zom-betatersteralt, 2023-08-09:
@Proto-App hi its me, please put the example by under this name: zom1proto

zom-betatersteralt, 2023-08-09:
since its my original username planned for the google signin

closed 2023-09-14 by Proto-App