Proto-App/Proto-Android #271: Questions about updates

Jurgen2003, 2023-08-12:
Hey @Proto-App thanks for the nice feature of update that time setting and theme for all circuit. I am. Curious when the power supply for opamp and sub circuit and also that import of circuit. In new circuit so you can build a one piece circuit and add to another. It would be nice feature if all goes well they are already on julie list for it but wonder when they would come.  For the rest. Fantastic app and Fantastic help here 
 Greetings. Jurgen

Proto-App, 2023-08-21:
Hello @Jurgen2003 
Circuit import is almost ready. Sure, we will add option to flip inputs/otputs in JS ic.

Jurgen2003, 2023-08-24:
@Proto-App to hear that this is in the planning I like to see it back like this as the power supply for the opamp. The import ( find it most convenient) sub circuit and java ic turnable  
 Greets  jurgen