Proto-App/Proto-Android #275: Bug

Yunhanar, 2023-08-22:
[image]

Proto-App, 2023-08-22:
@Yunhanar 
Thank you for report. Could you share your device model and android version?

Yunhanar, 2023-08-22:
Device : Vivo Y12s
Android version 11

Proto-App, 2023-08-22:
@Yunhanar 
Could you share one of the items from PROTO or whole workspace?

Yunhanar, 2023-08-23:
Like this? Sorry i cant upload a videos, gmail prevented me.

Proto-App, 2023-08-23:
You can tap on "Export workspace" and share link with us.

Yunhanar, 2023-08-23:
https://www.protosimulator.com/share?workspace=workspace_14ac8dc9-712b-4181-aac6-9d70b3d65aad_1692788815114

Proto-App, 2023-09-05:
We can't reproduce it on our devices. It must be connected to your vivo device. It has specific GPU MCU (PowerVR) and probably it causes this issue. We can't fix it.

closed 2023-09-05 by Proto-App