Proto-App/Proto-Android #278: Cannot start advance scope

horiz00n, 2023-09-05:
I've bought the advance scope addon, but I can't activate this feature, please help

[image]
[image]

Proto-App, 2023-09-05:
Hello @horiz00n 
Very strange issue. Could you uninstall and install PROTO once again and check this addon?

Proto-App, 2023-09-05:
@horiz00n any news in your case?

horiz00n, 2023-09-05:
Do I need to buy this addon again after reinstalling?

Proto-App, 2023-09-05:
No, it should available. Each item is connected to your Google account.

closed 2023-10-31 by Proto-App