Proto-App/Proto-Android #284: Add new circuit parts/feature

Amonia69420, 2023-10-10:
yeah, proto, can you add these:

- IGBT transistor
- changing the current needed to change relay's state (default 20mA, changeable up to 5A or something else)
- SRAM (digital)
- add option to add 2 wires at OpAmp
- add another optocoupler (example: 4N35)
- capacitor maximum volatge (changeable, can set true or false)
- resistor maximum power (changeable, can set true or false)
- PTC thermistor
- SR flip flop

Amonia69420, 2023-10-10:
c

Amonia69420, 2023-10-10:
> yeah, proto, can you add these:
>
> * IGBT transistor
> * changing the current needed to change relay's state (default 20mA, changeable up to 5A or something else)
> * SRAM (digital)
> * add option to add 2 [...]

- variable capacitor and variable inductor

Proto-App, 2023-10-10:
Hello @Amonia69420  
Thank you for feedback.

krotow, 2024-05-12:
AC input optocoupler with NPN transistor output (has input LEDs in both direction, real components are LTV-814, PC814, FOD814 etc.). I use these IRL.