Proto-App/Proto-Android #288: New Example: L298N

Amonia69420, 2023-10-21:
L298N is a motor driver that can control the direction of the motor, and speed of the motor. The motor is replaced by an inductor.
image:
[image]
link : https://www.protosimulator.com/share?circuit=L298N_1697859676351

Proto-App, 2023-10-21:
Thank you for this example! It will be present in version 1.26.0

Amonia69420, 2023-10-24:
wait, that's wrong image, this is the revision & the link:
[image]
https://www.protosimulator.com/share?circuit=Dual_H_Bridge_motor_1698108322266

Amonia69420, 2023-12-02:
thanks for adding that to proto

closed 2023-12-03 by Proto-App