Proto-App/Proto-Android #297: Frequency less than 1Hz

akeshma, 2023-11-27:
Please add Frequency less than 1Hz  in the components pluse and square signals

akeshma, 2023-12-30:
Thanks to add this feature

closed 2023-12-30 by akeshma

SzinNapalm, 2024-01-24:
I was wondering if you can add a speaker symbol to the  component list, and add a feature that allows the wave forms to be audible cuz I'm designing guitar pedals & synth hardware I have only just begun on my first one and I am loving this app PROTO is amazing if I were soldering this I would have had to take it apart so many times but with Proto I just hit delete component and reattach the wires a presto.   It would be great to get a few additional components like different transistors that are used in pedals and synth they really are the key to modern sound. Thanks