Proto-App/Proto-Android #300: Convergence failure even in lower timestep

Dreyfus0422, 2023-12-18:
Can't fix "Convergence failed, try to lower simulation time step." error

Proto-App, 2023-12-18:
Hello @Dreyfus0422 

Could you share your circuit?

Dreyfus0422, 2023-12-28:
See photo.Sent from my Huawei tablet-------- Original message --------From: PROTO Team ***@***.***>Date: Mon, 18 Dec 2023, 3:51 pmTo: Proto-App/Proto-Android ***@***.***>Cc: Dreyfus0422 ***@***.***>, Mention ***@***.***>Subject: Re: [Proto-App/Proto-Android] Convergence failure even in lower timestep (Issue #300)
Hello @Dreyfus0422
Could you share your circuit?

Proto-App, 2024-01-03:
@Dreyfus0422 
Sorry, but we can't see your photo here.

Dreyfus0422, 2024-06-23:
You review my photo.Sent from my Huawei tablet-------- Original message --------From: PROTO Team ***@***.***>Date: Thu, 4 Jan 2024, 12:19 amTo: Proto-App/Proto-Android ***@***.***>Cc: Dreyfus0422 ***@***.***>, Mention ***@***.***>Subject: Re: [Proto-App/Proto-Android] Convergence failure even in lower timestep (Issue #300)
@Dreyfus0422
Sorry, but we can't see your photo here.

piimah, 2026-01-11:
here on my s23+

https://github.com/user-attachments/assets/1c7fad11-e7ad-41a1-a6a9-eab90bd0c2ba

closed 2026-02-11 by Dreyfus0422