Proto-App/Proto-Android #302: A suggestion for new components

Jurgen2003, 2023-12-23:
Dear @Proto-App 
 A suggestion for new components 
   A 12 or 16 input and output javascript ic and a flatcabale that with 1 line and wire numbers the wires take less space. Until 1 other 5 signals can verzeten instead of 5 cores needed hear julie opinion and what we will see back 
Greetings jurgen

Proto-App, 2024-01-03:
Hello @Jurgen2003 
Could you explain it once again (we see words in your native language). Thank you.

Jurgen2003, 2024-01-06:
[image]

Dear @Proto-App
A 12 or 16 input and output javascript ic and a flatcabale see picture 
Such a cable might be easier to reduce the de. Quantity of wires  
Greetings jurgen

Jurgen2003, 2024-01-26:
Dear @Proto-App is it possible to use javascript if there is 0 to 10 volts. Negative voltage on an input there is a 0 to 10 volt positive from it. 
  And javascript ic with more inputs and outputs is that possible in a next version 
 Greetings jurgen

Vertaald met DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

krotow, 2024-05-12:
I guess he want to have a bus. This issue seems a double of #247.

Jurgen2003, 2024-05-12:
Dear @Proto-App 
  What I mean and would like are 2 things 
 1 .  Now if you draw 1 line/wire between 2 components for e.g. power supply can that line. 1 function. So for every function/communication there must be a line/wire so if an ic has 8 inputs and 8 outputs that ic needs 16 lines/wires that will be a big network of lines/wires. Would it be possible to put a line/wire behind the button for the name of a component and at power supplies where the voltage is set. At that button to put a kind of different number so you put all. 8 inputs and all 8 outputs on 1 wire and you set in wire 1 to 8 or on the ic input/output that you choose that simply put a cables also have multiple. Wires/veins so that an ic needs less space for its wires/lines 
 
  2
  If number 1 succeeded it would also be nice to have bigger javascript ic. With 16 inputs  
 If can also 32 but 16 would be nice. I have circuits with. Multiple 8 inputs ic but.  Need to search in which script. It is programmed. So. That 1 script is a big base of the circuit where by the space is less needed and the app becomes even more fun to build bigger circuits
 Greetings jurgen