Proto-App/Proto-Android #305: Here are Some Basic and important suggestions for improvements:

HoseinAgaei2003, 2023-12-25:
1-Time step, simulation speed, and current speed of circuits numerically be exactly changeable ( be changeable by typing exact amounts ).

2-All locations of selected area's components be movable together, Instead of having us to move them one by one and be tired and also lose time, especially in big circuits, O Dears.

3-Please add exactly DC Power supply itself,
That is separate from the (Battery) symbol as the DC power supply in components of this app, dears🙏🏻.

4-Please add Phototransistor.

🌹🙏🏻🌹

Proto-App, 2023-12-25:
Duplicate: https://github.com/Proto-App/Proto-Android/issues/304

closed 2023-12-25 by Proto-App

HoseinAgaei2003, 2023-12-25:
> Duplicate: #305

 Hi, I am Sorry about #305 , these suggestions were sent twice, it was unintentional, I again demand your forgiveness. I have Don't be tired and be strong wish for you, these improvements are for our benefit and especially for you. I hope you take full advantage of this improvement offers 💪🏻🙌🏻🌹 .

Just is #304 , just have #304 .