Proto-App/Proto-Android #307: Can't use delay in JavaScript IC

elesbb, 2024-01-03:
Hello, I bought the JavaScript IC pack thinking I could use it like a microcontroller (arduino), but I cannot. 

Is there any way you can add an Arduino pack and/or make it so the JavaScript IC can do more complex things like having a while loop with delays?

Proto-App, 2024-01-03:
Hello @elesbb 
Sorry but we don't support any Arduino library. You can't make conventional delay function because simulation engine executes your code (it should be as fast as possible). But you can use "global" variables and introduce delay like this:

```js
globa.declareDouble("simulationDelay", 0.0); //one time init at first execution

//then use it in code with if statement:

let accumulatedDelay = simulationDelay.get() + env.getTimeStep(); //get old value and add next simulation time step
if (accumulatedDelay >= 0.5) {
  //do something after 0.5 sec
  simulationDelay.set(0.0); //reset delay
}

elesbb, 2024-01-03:
> Hello @elesbb Sorry but we don't support any Arduino library. You can't make conventional delay function because simulation engine executes your code (it should be as fast as possible). But you can use "global" variables and introduce delay like this:

I see. That makes sense. Thanks.

Is there any possible plans for implementing an Arduino Uno as a pack? Something like wokwi.com but in your app. It would be very awesome to be able to do that.

I think it should be relatively straight forward to implement. You could change the evaluations of the components to essentially run in the "loop" function of the Arduino at whatever time step the Arduino clock speed is running at.

Just an idea, would be an insanely awesome addition to your incredible app.

Proto-App, 2024-01-03:
Definitely we want to add Arduino component 🙂and wokwi is perfect example. But on the way is global online library in our app (70% is done) and future iOS version because we have many requests about it. Finally we will add Arduino (maybe MSP430 too) but it will take some time.

elesbb, 2024-01-03:
Psh who cares about iOS, Android is superior lol. But thanks for the timeline!!! Can't wait till future releases!

If you need a beta tester, I am very good with coding and debugging and would gladly help out!

Proto-App, 2024-01-03:
Sure, we will notify you about alpha tests.

closed 2024-09-02 by Proto-App