Proto-App/Proto-Android #308: How to set and modify external names of custom IC

Sureshot14, 2024-01-04:
Hello, I am trying to find out how to change the external names of the custom ICs. Specifically I am trying to change the 8 pin IC. While I am not 100% familiar with Javascript, I am aware that this may not be my limit, and seems to be with Proto itself. I am trying to rename the INs and OUTs to Pin 1-8. I am also wondering if the input and outputs work realistically to the flow of electricity, as in does a in act as a positive and out as a negitive. In a real circuit out would have electrons coming from it going to the in. If this is how it works then I would be assigning the IC as follows: IN0 = pin1, IN1 = pin2, IN2 = pin3, IN3 = pin4, OUT3 = pin5, OUT2 = pin6, OUT1 = pin7, OUT0 = pin8. Then I would be assigning the pins with the needed input and output as followes: pin1 = in, pin2 =out, pin3 = in, pin4 = out, pin5 = out, pin6 = out, pin7 = in, pin8 = out. I am not sure how to display anything other than what is on the IC though, and don't know how to change any of the ins to outs and vise versa. I want it to display the pins if possible, and I am wondering how to assign the external pins to certain variables. It is not specific and clear on this. I think it may be that you would assign them as follows: in0 = "your variable", and all of them the same way. If anyone can help me or give insight I would appreciate it.

Proto-App, 2024-01-04:
Hello @Sureshot14 
Sorry but we don't support pin renaming. Do you want to rename only visible text or maybe assign input value to output?

Proto-App, 2024-01-04:
By the way you can't assign anything to in0...inN because you can only read  from input pins. You can write value to out0...outN pins.

Sureshot14, 2024-01-04:
Both, it's not really a custom IC if you are locked at all the ICs having fixed inputs and outputs. The 8 pin IC is perfect for what I need but I need 5 outputs and 3 inputs. That is if it supports a realistic circuit, otherwise I would need the inverse, 5 inputs and 3 outputs. I want to be able to name the external text and change all the OUT and IN to pins 1-8. I don't think any of this would be hard to implement. I am ultimately trying to recreate the THAT 1512 IC. It has 8 pins with 5 outputs and 3 inputs. Four of those are connected to 2 different dc circuits, positive and negative/ground, and one connected to ground. Two are a self contained gain control, and the last is a output from a microphone.

Proto-App, 2024-01-04:
Thank you for feedback. Yes custom IC is a blocker in your case. We can at least provide version with more inputs and outputs (14PIN) or add ability to re-program inputs <-> outputs. You can expect one of the mentioned solutions in next release.

Sureshot14, 2024-01-04:
Giving a 14 pin wouldn't really fix my issue, but being able to change the inputs to outputs and vise versa would. I want it to be as clean as possible.

Proto-App, 2024-01-09:
@Sureshot14 
We don't understand your request. Custom IC has 8 inputs and 8 outputs.  You need only 5 outputs and 3 inputs. It looks like 8in-8out version should satisfy your requirements.

Sureshot14, 2024-01-09:
The larger ICs are bigger and take more space, it would be a lot better if you could change the inputs and outputs. It would also allow circuits to be more organized. So I am asking if you could update the app to allow you to ICs to be coded so that you can assign names and change inputs and outputs.