Proto-App/Proto-Android #321: a question and ideas / tips

Jurgen2003, 2024-02-06:
dear @Proto-App 

thanks for the latest update with new options  
 reading out sensors with java how does that work 
 and is it possible to put an output in high at for example 5 volts and low at 2.5 volts 
 so out0.setState( in0.getvoltage() >5.0 ) to high  
and out0.setState( in0.getvoltage() < 2.5 ) to low 
  
  just like the schimmit trigger voltage uppr and voltage lower 
for high and low  

  and also with me the app sometimes cracks at large circuits and restarts 
 after a few seconds 

otherwise still a perfect app

Proto-App, 2024-02-06:
Hello @Jurgen2003 
Thank you for feedback! In next release we will provide one or two examples with sensor reading. You will be able to learn how to use it in real code. As always please check "Help" section in code editor. It has basic how-to example.

Jurgen2003, 2024-02-06:
Dear @Proto-App   
 Glad to see preimages in the next version 
 Is it also possible to extend the java script package with larger ic like in the doubling  
  1 ,2 ,4 ,8 16 ,32  
 And that the javascript ic can be run  
 That would be a good improvement 
 Like to hear your reactions

Proto-App, 2024-02-12:
Duplicate https://github.com/Proto-App/Proto-Android/issues/322

closed 2024-02-12 by Proto-App