Proto-App/Proto-Android #324: Help

Jurgen2003, 2024-02-08:
What's wrong with this script 
  It is meant to give output 0 and1 a clamp but with other bounds what am I doing wrong and is it possible  

function clamp(num, min, max) {
     return Math.min(Math.max(num, min), max);
}

function setupTerminals() {
    let outVoltage1 = clamp1(in0.getVoltage() + in1.getVoltage(), 2, 12)    //clamp out voltage between 2v - 12v
    out0.setLowVoltage(outVoltage1);

  function setupTerminals() {
    let outVoltage2 = clamp2(in2.getVoltage() + in3.getVoltage(), 4, 15)    //clamp out voltage between 4v - 15v
    out0.setLowVoltage(outVoltage1);
    out1.setLowVoltage(in0.getVoltage() - in1.getVoltage()); // out1 voltage = input 0 - input 1
}

function driveOutputs() {
    //turn output to high state 
    
  
}

//code loop executes both functions
setupTerminals();
driveOutputs();

[image]

Proto-App, 2024-02-08:
Hello @Jurgen2003 
Could you share link to your circuit?

Jurgen2003, 2024-02-09:
The idea is to have multiple clamps in 1 ic with more outputs and multiple boundaries so. Output 0 limits  
Output 1 other limits   
 And so on 
   See the ic stacked and that in 1 ic 

https://www.protosimulator.com/share?circuit=Circuit_3_1707479112804

Proto-App, 2024-02-09:
Here is modified version: https://www.protosimulator.com/share?circuit=Circuit_3_1707484196607

Jurgen2003, 2024-02-09:
Dear@Proto-App  
[image]

I get this message with an open from a shared link 
Also julie's  
With wifi and 4g 
 What could it be

closed 2024-02-12 by Proto-App