Proto-App/Proto-Android #329: Workspace import issue

borghim, 2024-03-03:
Hi! I have problem importing my workspace from old device Samsung ZFlip3 to the new one which is Samsung ZFlip5. Same Android version 14 and same app version 1.27.0. 
Workspace export link is https://www.protosimulator.com/share?workspace=workspace_d0cbed77-2884-4d02-bed7-835e1050e425_1709454682582

When I open the app in the new device and click on this link I have an error reported in the image attached.
Thanks 
[image]

Proto-App, 2024-03-03:
Hello @borghim 
On our device it works fine (there are 5 circuits).

Could you check "Supported web addresses" setting? You can find it in following steps (Samsung Galaxy Tab): 
Android settings -> Apps -> PROTO -> Set as default (Open supported links option should be turned on) -> Supported web addresses -> www.protosimulator.com should be turned on.

borghim, 2024-03-03:
Thanks for prompt reply!
Yes, 5 circuits..glad that it works for you.
The settings you are talking about I have changed already before otherwise the link was taken as a webpage. In my case the link opens the app but nothing is imported due to this unknown error..
[image]
[image]

Proto-App, 2024-03-03:
Ok, very strange. Could you try to import one of your circuits: https://www.protosimulator.com/share?circuit=Passaggi_0_Trifase_1709457559391 ? If it won't work then we need to add more logs to PROTO to debug this issue.

borghim, 2024-03-03:
Unfortunately even one by one doesn't work for me. Two more info: 
1) I have succesfully imported the whole workspace on another phone but on this phone I don't own any addons and so I've missing parts.
2) I have tried to import the workspace on the old phone Zflip3, the phone used to generate the export link, but I have the same error.
I have uninstalled and installed back the app again but same thing..
Now, since the export link works and the problem seems to be only import, I will reset old phone to factory because I have sold it.
Thanks for your effort, I'm available for further test with the new phone.
Let me know, thanks

Proto-App, 2024-03-03:
Thank you for your feedback. Addons are not required to import/export. Can we prepare special version for you with more logs to debug this issue? It will be available for you via "Internal test track", we need some time to prepare it.

Proto-App, 2024-03-03:
Yes, you need addons if you have restricted components. All addons are tied to Google account (this is default behavior in Android). We would like to solve import issue because it is basic functionality and it should work :) We will prepare new version in next few days and will back to you.

borghim, 2024-03-03:
👍

Jurgen2003, 2024-03-08:
I also keep getting errors with importing

Proto-App, 2024-03-08:
Yes, we know. Tomorrow we will prepare test version. Of course you can participate with it :+1:

Proto-App, 2024-03-09:
@Jurgen2003 @borghim 
Could you share Gmail account for testing? We need them to give you access to new version.

Jurgen2003, 2024-03-09:
Dear proto  
My email is jurgenloosduinen@gmail.com
Thanks in advance

Proto-App, 2024-03-09:
Thanks, we will write you an email about further steps.

Jurgen2003, 2024-03-09:
[image]

What is this message

borghim, 2024-03-09:
@Proto-App 
my email is miborghi@gmail.com 
Thanks in advance

Proto-App, 2024-03-09:
@Jurgen2003 
Now it should be available

Proto-App, 2024-03-09:
@Jurgen2003 
Do you have device with Android 14?

Jurgen2003, 2024-03-09:
Yes i have. Andriod 14   6.0

Proto-App, 2024-03-09:
@Jurgen2003 @borghim 
Ok, probably this bug is connected with Android 14. On Monday we will provide fix in test version.

Jurgen2003, 2024-03-09:
Okay we'll see  
 If in the test version the problem is fixed it then via github. Report it ?

Proto-App, 2024-03-09:
@Jurgen2003 @borghim 
Fixed version should be available for you :wink:

Jurgen2003, 2024-03-10:
@Proto-App 
The import problem in andriod 14 in with me. Resolved after updating to 1.27.1   
Greetings jurgen

borghim, 2024-03-10:
@Proto-App 
I confirm that also for me the issue has been solved since my workspace has been succesfully imported. Great job! 
Shall we uninstall test/debug version and install regular version from play store or not yet?
Thanks!

Proto-App, 2024-03-10:
@Jurgen2003 @borghim 
Great! We will release new version with fix on Monday in rollout mode. It should be available for everyone at the end of the week. You should uninstall test version and install official to receive future update.

Proto-App, 2024-03-14:
@Jurgen2003 @borghim 
Fixed in 1.28.0

Jurgen2003, 2024-03-14:
@Proto-App 
Thanks for fixing the problem version 1.28 installed and it works
Greets jurgen

closed 2024-03-15 by Proto-App