Proto-App/Proto-Android #330: Regarding addon

hemanthiiser, 2024-03-03:
I purchased analog addaon package and when I updated the app, my addon package was lost. It's asking to purchase again.
Please resolve my issue. 
[image]
[image]

Proto-App, 2024-03-03:
Hello @hemanthiiser 
As we can see you have available Misc Pack, not Analog.

hemanthiiser, 2024-03-03:
Thanks for correcting. It's misc pack. Still that also not showing as owner for me. I lost the access. 

[image]

hemanthiiser, 2024-03-03:
> Hello @hemanthiiser As we can see you have available Misc Pack, not Analog.

Thanks for correcting, still misc pack also I lost.

Proto-App, 2024-03-03:
Could you share order id (GPA-xxx)? Or account email address?

hemanthiiser, 2024-03-04:
Email id: hemanthfirms@gmail.com

UPI transaction I'd: 406406949403
[image]

hemanthiiser, 2024-03-04:
I was just about to purchase the bundle pack. But by looking this for just one update, It was risky to purchase. 
Pls also give a solution.

Proto-App, 2024-03-04:
@hemanthiiser Could you clear app data or uninstall PROTO and install once again?

hemanthiiser, 2024-03-05:
Thank you for your email. The issue is resolved. I can able to access now.

hemanthiiser, 2024-03-05:
Will this problem again raises if new update comes?

Proto-App, 2024-03-05:
@hemanthiiser Good news! Update should not cause any issues with addon. Do you remember this old version numer?

hemanthiiser, 2024-03-05:
1.26.0 I think.

Proto-App, 2024-03-08:
Ok, thank you for info.

closed 2024-03-08 by Proto-App